teemuatlut/TMC2208Stepper #1: library does not work on x86

pirat777knj, 2017-12-24:
Hey.
I decided to put 2208 on RADDS, but errors are made when compiling.
The compiler warns that the library does not work on x86 (((

`ПРЕДУПРЕЖДЕНИЕ: библиотека TMC2208Stepper должна запускаться на архитектурах (avr) и может быть несовместима с вашей платой на архитектуре (sam).
sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type

   #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \

                                        ^

sketch\src\core\stepper\stepper_indirection.cpp:255:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(X);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperX_serial' was not declared in this scope

                                        TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)

                                                                    ^

sketch\src\core\stepper\stepper_indirection.cpp:255:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(X);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type

   #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \

                                        ^

sketch\src\core\stepper\stepper_indirection.cpp:269:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(Y);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperY_serial' was not declared in this scope

                                        TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)

                                                                    ^

sketch\src\core\stepper\stepper_indirection.cpp:269:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(Y);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type

   #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \

                                        ^

sketch\src\core\stepper\stepper_indirection.cpp:276:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(Y2);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperY2_serial' was not declared in this scope

                                        TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)

                                                                    ^

sketch\src\core\stepper\stepper_indirection.cpp:276:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(Y2);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type

   #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \

                                        ^

sketch\src\core\stepper\stepper_indirection.cpp:283:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(Z);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperZ_serial' was not declared in this scope

                                        TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)

                                                                    ^

sketch\src\core\stepper\stepper_indirection.cpp:283:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(Z);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:247:40: error: 'SoftwareSerial' does not name a type

   #define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial stepper##ST##_serial = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \

                                        ^

sketch\src\core\stepper\stepper_indirection.cpp:297:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(E0);

       ^

sketch\src\core\stepper\stepper_indirection.cpp:248:68: error: 'stepperE0_serial' was not declared in this scope

                                        TMC2208Stepper stepper##ST(&stepper##ST##_serial, ST##_SERIAL_RX_PIN > -1)

                                                                    ^

sketch\src\core\stepper\stepper_indirection.cpp:297:7: note: in expansion of macro '_TMC2208_DEFINE_SOFTWARE'

       _TMC2208_DEFINE_SOFTWARE(E0);

       ^`

teemuatlut, 2017-12-24:
The SoftwareSerial library is incompatible with Arduino Due and TMC2208Stepper is depending on that library.

pirat777knj, 2017-12-24:
Very bad((((
The question remains: How to connect the driver?

teemuatlut, 2017-12-24:
It might be possible to use the hardware serial for a few drivers but you first need to comment out some code in Marlin and the library. A better solution is to extend the software serial library to support Due or to find an alternative library.

pirat777knj, 2017-12-25:
Something sad (((

closed 2018-01-21 by teemuatlut

weed2all, 2018-10-11:
So any workaround this? I'm interested to have my tmc2208 uart mode on arduino due + radds...
Any news?