teemuatlut/TMC2208Stepper #5: RX lines on MKS not working?

melvinisken, 2018-01-21:
I did some further testing and I found another strange issue, that I cannot resolve at the moment. It's not directly related to the library but I didn't find another place to post this. On an MKS 1.4 board, I installed 4 TMC2208 (X,Y,Z,E0). The RX/TX pins are connected to AUX 2. I now have the issue, that I can write to all of the modules (I am able to swith turning direction via driver.shaft()) but if I try to read the drive status, only X and E0 return the 32 bits value, Y and Z return just 0. I switched modules and the UART cables, everything works fine using the RX/TX pins used for X and E0. So it must be something with pins 40 and 42 I guess. Maybe they are wired differently on the MKS board? Can someone try to reproduce this? I used the pins from pins_RAMPS.h in Marlin. Here's my test sketch:

```
// Author Teemu Mäntykallio, 2017-04-07

// Define pins

//X
/*#define EN_PIN    38                               // LOW: Driver enabled. HIGH: Driver disabled
#define DIR_PIN   55                              // Set stepping direction
#define STEP_PIN  54                              // Step on rising edge
#define TX_PIN    59                              // SoftwareSerial
#define RX_PIN    63                              // 
*/
//Y
#define EN_PIN    56 				// LOW: Driver enabled. HIGH: Driver disabled
#define DIR_PIN   61 				// Set stepping direction
#define STEP_PIN  60 				// Step on rising edge
#define TX_PIN 	  64				// SoftwareSerial
#define RX_PIN    40				// 

//Z
/*#define EN_PIN    62                               // LOW: Driver enabled. HIGH: Driver disabled
#define DIR_PIN   48                              // Set stepping direction
#define STEP_PIN  46                              // Step on rising edge
#define TX_PIN    44                              // SoftwareSerial
#define RX_PIN    42                              // 
*/
//E0
/*#define EN_PIN    24                               // LOW: Driver enabled. HIGH: Driver disabled
#define DIR_PIN   28                              // Set stepping direction
#define STEP_PIN  26                              // Step on rising edge
#define TX_PIN    66                              // SoftwareSerial
#define RX_PIN    65                              // 
*/
0
#include <SoftwareSerial.h>
SoftwareSerial tmc_sw = SoftwareSerial(RX_PIN, TX_PIN);

#include <TMC2208Stepper.h>				// Include library

#include <TMC2208Stepper_REGDEFS.h>

TMC2208Stepper driver = TMC2208Stepper(&tmc_sw);	// Create driver that uses SoftwareSerial for communication

void setup() {
  Serial.begin(115200);
  tmc_sw.begin(115200);			// Init used serial port
  while (!tmc_sw);					// Wait for port to be ready

  // Prepare pins
  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);

  driver.pdn_disable(1);				// Use PDN/UART pin for communication
  driver.rms_current(500, 0.11, 0.5);		// Set driver current = 500mA, RSENSE = 0.11 and 0.5 multiplier for hold current.
  driver.en_spreadCycle(false);
  driver.shaft(false);
  //driver.toff(0x2);						// Enable driver

  digitalWrite(13, LOW);					// Enable driver

  uint32_t data;
  driver.DRV_STATUS(&data);
  Serial.println(data, BIN);
}

void loop() {
  digitalWrite(STEP_PIN, !digitalRead(STEP_PIN)); // Step

  delay(10);
}
```

brisma, 2018-01-22:
Same issue, I've a MKS Gen 1.4 board and five TMC2208.

I can write the firmware by FTDI Usb programmer correctly (I switched to SpreadCycle on E0 via usb), but I cannot communicate with the Marlin Firmware with UART. No matter which pins I'm using, I tried many combinations, no one is able to communicate with the SoftwareSerial :(

melvinisken, 2018-01-23:
How did you wire your modules? The communication is working at my board. The strange thing is that only 2 out of 4 pin pairs are working and the other 2 don't. From a hardware perspective I can communicate to all modules, but only two at once (the two where the working UART pins are connected to). Which Marlin version are you using?

brisma, 2018-01-23:
Hi melvinisken,
I found in my case that the problem was not related to the wiring of the driver, but was a bug in the Marlin firmware.

The UART communication was working correctly only on HardwareSerial and not in SoftwareSerial... because the begin method of the serial communication was missing in case of SoftwareSerial :)

I just submitted a PR on Marlin to fix this issue.

melvinisken, 2018-01-24:
Ok, cool, I was not aware of this. I only tested the UART communication with the example code and thought Marlin should work as well. I will include the bugfix and try again tonight. But I guess that I will still have two working lines and two broken ones. Maybe my MKS board is defective.

closed 2018-05-08 by melvinisken