teemuatlut/TMC2208Stepper #15: 'class TMC2208Stepper' has no member named 'hysterisis_start'

oliclarke, 2018-03-19:
Just updated to v3 and getting this error on compile. 

Arduino: 1.8.4 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\stepper_indirection.cpp: In function 'void tmc2208_init(TMC2208Stepper&, uint16_t, uint32_t, float)':

stepper_indirection.cpp:391: error: 'class TMC2208Stepper' has no member named 'hysterisis_start'

     st.hysterisis_start(3);

        ^

stepper_indirection.cpp:392: error: 'class TMC2208Stepper' has no member named 'hysterisis_end'

     st.hysterisis_end(2);

        ^

exit status 1
'class TMC2208Stepper' has no member named 'hysterisis_start'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

teemuatlut, 2018-03-20:
You need to either upgrade to bugfix-1.1.x branch or downgrade the library. However I will be making a release to restore backward compatibility for those who for one reason or another need to run an older version of Marlin.

oliclarke, 2018-03-20:
Currently running bugfix 1.1.x i'll grab the most recent and try again.

closed 2018-03-24 by oliclarke