teemuatlut/TMC2208Stepper #17: Compile error TMC2208 with Marlin (Undefined reference)

Bumperchaser, 2018-05-03:
Hi,

I blew my previous board and stepper drivers so decided to upgrade somewhat.

I've run into a compile error with my setup that I can't work out.

I have TMC2208 fitted to X, Y, Z and E0 fitted to a MKS GEN L board.

I tried to compile with "HAVE_TMC2208" enabled and have added the TMC2208Stepper.h and TMC2208StepperrREGDEFS.h files to the sketch but the compiler returns the following:

C:\Users\johnh\AppData\Local\Temp\cconGa2O.ltrans0.ltrans.o: In function `process_parsed_command()':
cconGa2O.ltrans0.o:(.text+0x2c52): undefined reference to `TMC2208Stepper::setCurrent(unsigned int, float, float)'
cconGa2O.ltrans0.o:(.text+0x2cdc): undefined reference to `TMC2208Stepper::clear_otpw()'
cconGa2O.ltrans0.o:(.text+0x2cf8): undefined reference to `TMC2208Stepper::clear_otpw()'
cconGa2O.ltrans0.o:(.text+0x2d14): undefined reference to `TMC2208Stepper::clear_otpw()'
cconGa2O.ltrans0.o:(.text+0x2d26): undefined reference to `TMC2208Stepper::clear_otpw()'

C:\Users\johnh\AppData\Local\Temp\cconGa2O.ltrans5.ltrans.o: In function `MarlinSettings::report(bool)':
cconGa2O.ltrans5.o:(.text+0x684): undefined reference to `TMC2208Stepper::getCurrent()'
cconGa2O.ltrans5.o:(.text+0x69e): undefined reference to `TMC2208Stepper::getCurrent()'
cconGa2O.ltrans5.o:(.text+0x6b8): undefined reference to `TMC2208Stepper::getCurrent()'
cconGa2O.ltrans5.o:(.text+0x6d2): undefined reference to `TMC2208Stepper::getCurrent()'

C:\Users\johnh\AppData\Local\Temp\cconGa2O.ltrans5.ltrans.o: In function `void tmc_report_otpw<TMC2208Stepper>(TMC2208Stepper&, TMC_AxisEnum)':
cconGa2O.ltrans5.o:(.text+0xc92): undefined reference to `TMC2208Stepper::getOTPW()'

C:\Users\johnh\AppData\Local\Temp\cconGa2O.ltrans5.ltrans.o: In function `void tmc_get_current<TMC2208Stepper>(TMC2208Stepper&, TMC_AxisEnum)':
cconGa2O.ltrans5.o:(.text+0xcfe): undefined reference to `TMC2208Stepper::getCurrent()'

C:\Users\johnh\AppData\Local\Temp\cconGa2O.ltrans7.ltrans.o: In function `tmc2208_init(TMC2208Stepper&, unsigned int, unsigned int, unsigned long, float) [clone .constprop.8]':
cconGa2O.ltrans7.o:(.text+0x4ac): undefined reference to `TMC2208Stepper::pdn_disable(bool)'
cconGa2O.ltrans7.o:(.text+0x4b4): undefined reference to `TMC2208Stepper::mstep_reg_select(bool)'
cconGa2O.ltrans7.o:(.text+0x4bc): undefined reference to `TMC2208Stepper::I_scale_analog(bool)'
cconGa2O.ltrans7.o:(.text+0x4da): undefined reference to `TMC2208Stepper::rms_current(unsigned int, float, float)'
cconGa2O.ltrans7.o:(.text+0x4e4): undefined reference to `TMC2208Stepper::microsteps(unsigned int)'
cconGa2O.ltrans7.o:(.text+0x4ec): undefined reference to `TMC2208Stepper::blank_time(unsigned char)'
cconGa2O.ltrans7.o:(.text+0x4f4): undefined reference to `TMC2208Stepper::toff(unsigned char)'
cconGa2O.ltrans7.o:(.text+0x4fc): undefined reference to `TMC2208Stepper::intpol(bool)'
cconGa2O.ltrans7.o:(.text+0x50a): undefined reference to `TMC2208Stepper::TPOWERDOWN(unsigned long)'
cconGa2O.ltrans7.o:(.text+0x512): undefined reference to `TMC2208Stepper::hysteresis_start(unsigned char)'
cconGa2O.ltrans7.o:(.text+0x51a): undefined reference to `TMC2208Stepper::hysteresis_end(signed char)'
cconGa2O.ltrans7.o:(.text+0x522): undefined reference to `TMC2208Stepper::pwm_lim(unsigned char)'
cconGa2O.ltrans7.o:(.text+0x52a): undefined reference to `TMC2208Stepper::pwm_reg(unsigned char)'
cconGa2O.ltrans7.o:(.text+0x532): undefined reference to `TMC2208Stepper::pwm_autograd(bool)'
cconGa2O.ltrans7.o:(.text+0x53a): undefined reference to `TMC2208Stepper::pwm_autoscale(bool)'
cconGa2O.ltrans7.o:(.text+0x542): undefined reference to `TMC2208Stepper::pwm_freq(unsigned char)'
cconGa2O.ltrans7.o:(.text+0x54a): undefined reference to `TMC2208Stepper::pwm_grad(unsigned char)'
cconGa2O.ltrans7.o:(.text+0x552): undefined reference to `TMC2208Stepper::pwm_ofs(unsigned char)'
cconGa2O.ltrans7.o:(.text+0x55a): undefined reference to `TMC2208Stepper::en_spreadCycle(bool)'
cconGa2O.ltrans7.o:(.text+0x568): undefined reference to `TMC2208Stepper::GSTAT(unsigned long)'

C:\Users\johnh\AppData\Local\Temp\cconGa2O.ltrans7.ltrans.o: In function `global constructors keyed to 65535_0_G26_Mesh_Validation_Tool.cpp.o.3750':
cconGa2O.ltrans7.o:(.text.startup+0x18): undefined reference to `TMC2208Stepper::TMC2208Stepper(Stream*, bool)'
cconGa2O.ltrans7.o:(.text.startup+0x34): undefined reference to `TMC2208Stepper::TMC2208Stepper(Stream*, bool)'
cconGa2O.ltrans7.o:(.text.startup+0x50): undefined reference to `TMC2208Stepper::TMC2208Stepper(Stream*, bool)'
cconGa2O.ltrans7.o:(.text.startup+0x6c): undefined reference to `TMC2208Stepper::TMC2208Stepper(Stream*, bool)'

I am new to this and have no idea where to start to get to the bottom of this!

I have the sketch stored in E:, so why would it be looking at C: directories for compilation?

Please help!!

Bumperchaser, 2018-05-03:
When I do that, the compiler comes up with "No such file or directory"

teemuatlut, 2018-05-03:
Reinstall the library and try compiling the Simple example.

Bumperchaser, 2018-05-05:
Solved!
It seems there is a problem with the setup on my desktop. I installed Arduino and the libraries on my laptop and it compiled without any issues!
Thank you for the quick responses!

John

legalstuff, 2018-11-03:
I know it´s a long time to go.. but may anybody can help me.

Today, what is the best fit libary for the Fysecs TMC2208 or Biqu Bigtree TMC2208 (if there a difference)?
Wich Verson of Marlin is recomended?
Wich Version of Arduino  1.8.7 or 1.8.6?