teemuatlut/TMC2208Stepper #19: Multiple instances with SW serial read issue

ACE1046, 2018-08-21:
Thanks for library!
I'm trying to setup TMC2208 drivers on all axis on my RAMPS 1.4.
I'm connected all drivers with RX & TX, set correct pins in config, using software serial pins.
Problem is M122 command writes "0xC0:0C:00:00" for all axis even if I disconnect all pdn_uart pins except X axis. And I get "0x00:00:00:00" an all axis if I disconnect pdn_uart from X driver.
I wrote a simple sketch to check with. 

#include <TMC2208Stepper.h>             // Include library
TMC2208Stepper driver1 = TMC2208Stepper(63, 40, 1);  // Create driver and use
TMC2208Stepper driver2 = TMC2208Stepper(66, 44, 1);  // Create driver and use

void setup() {
  Serial.begin(115200);             // Init used serial port

  driver1.beginSerial(115200);
  driver2.beginSerial(115200);
}

void loop() {
  uint32_t g;
  driver1.GCONF(&g);
  Serial.println(g);
  delay(1000);
  driver2.GCONF(&g);
  Serial.println(g);
  delay(1000);
}

Real values a read from port, witch driver created first. If I disconnect wire for first driver, I get zeros for both.

Looks like changing this 2 lines in TMC2208Stepper::TMC2208Stepper fixes problem:
	//static SoftwareSerial mySWSerial = SoftwareSerial(SW_RX_pin, SW_TX_pin);
	//SWSerial = &mySWSerial;
	SoftwareSerial *mySWSerial = new SoftwareSerial(SW_RX_pin, SW_TX_pin);
	SWSerial = mySWSerial;

After this fix, I get correct (different) status for different axis in marlin. Disconnecting individual driver results in correct M122 details.

I'm just a n00b, may be I didn't understand something, sorry.

teemuatlut, 2018-08-21:
Alright thanks. If this works, it may be the answer to some of the error reports that were reported from Marlin users.

teemuatlut, 2018-08-21:
The fix is looking good on my end. Many thanks!
I'd still like to verify from a few sources that it's working now before creating a new release but the fix itself is online already.
I'll make sure to credit you if this goes through.