teemuatlut/TMC2208Stepper #20: UART on 32bit Re-Arm

oakdesign, 2018-08-24:
I'm going to receive a Re-Arm in the next few days, replacing my RAMPS Atmega2560 Setup. From my Research so far there shouldn't be an issue to run my TMC2130 in SPI, the question is about the TMC2208 I'm using as well. As the Re-Arm doesn't break out the usual TX / RX Pins 40,42,44 and 63 64 65 so SW_Serial can not be used. 
Could you share Information on how to possible set up and connect 2208 with Hardware Serial.

Regards Richard

weed2all, 2018-10-17:
Any further work on this...im kinda having same issue here but with arduino due and radds shield...i will love to se this working...im only planning to use X Y Z with tmc2208

teemuatlut, 2018-10-17:
On Re-Arm the software serial might work on different pins. Maybe on the ETH port, but I haven't tried this and I don't know if it would work.

DUE should work when used with hardware serial but SW Serial is not supported due to limited library availability.

simonedallai, 2018-12-07:
http://ustepper.com/store/ustepper-boards/26-42-ustepper-s-lite.html#/28-connector-dc_jack/31-stepper_motor-no_stepper_motor

teemuatlut, 2018-12-07:
Please make a new ticket as this one concerns Re-Arm. It's also a good practise to paste in the compiler error you'd like to report.

simonedallai, 2018-12-07:
Oh I 'm sorry!

necris38, 2019-02-01:
Hi ! I try too to declare my TMC2208 in UART mode in Marlin with my Re-arm/RAMPS board. Is it functionnal for somebody ?
TX/RX pin are descripted in "pins_RAMPS.h" but not in "pins_RAMPS_RE_ARM.h".
I don't know where to define my RX/TX pin...