teemuatlut/TMC2208Stepper #23: Function shaft not working to change motor direction

YannickRiouRobotics, 2018-11-10:
I develop on Wemos Lolin32 lite board and manage to drive the TMC2208 driver. I can communicate with the driver but I can't manage to make the shaft function work. It should make the motor change direction but it doesn't on my setup. 

Here is the code that I use to make the direction change : 
`bool motDir = true;
// Main loop
void loop()
{
    for (int i = 0; i < 40; i++)
    {
      digitalWrite(STEP_PIN, HIGH);
      delayMicroseconds(300); 
      digitalWrite(STEP_PIN, LOW);
      delayMicroseconds(300); 
    }
    driver.shaft(motDir);
    motDir=!motDir;
  
  }`

wehibeg, 2018-11-26:
I do have the same problem setting up on Atmega328p MCU

Hopefully someone will put together the solution 

Thanks for posting

wehibeg, 2018-12-01:
Spent sometime trying to move a different direction and no luck.   it seems  not much support or resource available on TMC2208.  

Too bad!

teemuatlut, 2018-12-01:
Tried it and it works.

```cpp
// Author Teemu Mäntykallio, 2017-04-07

// Define pins
#define EN_PIN    38  // LOW: Driver enabled. HIGH: Driver disabled
#define STEP_PIN  54  // Step on rising edge
#define RX_PIN    63  // SoftwareSerial pins
#define TX_PIN    40  //

#include <TMC2208Stepper.h>

// Create driver that uses SoftwareSerial for communication
TMC2208Stepper driver = TMC2208Stepper(RX_PIN, TX_PIN);

bool dir = true;

void setup() {
  Serial.begin(250000);
  while(!Serial);
  driver.beginSerial(115200);
  // Push at the start of setting up the driver resets the register to default
  driver.push();
  // Prepare pins
  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);

  driver.pdn_disable(true);     // Use PDN/UART pin for communication
  driver.I_scale_analog(false); // Use internal voltage reference
  driver.rms_current(500);      // Set driver current = 500mA, 0.5 multiplier for hold current and RSENSE = 0.11.
  driver.toff(2);               // Enable driver in software

  uint32_t drv_status;
  driver.DRV_STATUS(&drv_status);
  Serial.print("drv_status=");
  Serial.println(drv_status, HEX);

  switch (driver.test_connection()) {
    case 1: Serial.println("Connection error: F");
    case 2: Serial.println("Connection error: 0");
    default: Serial.println("Connection OK");
  }

  digitalWrite(EN_PIN, LOW);    // Enable driver in hardware
}

void loop() {
  digitalWrite(STEP_PIN, HIGH);
  delay(1);
  digitalWrite(STEP_PIN, LOW);
  delay(1);
  uint32_t ms = millis();
  static uint32_t last_time = 0;
  if ((ms - last_time) > 2000) {
    if (dir) {
      Serial.println("Dir -> 0");
      driver.shaft(0);
    } else {
      Serial.println("Dir -> 1");
      driver.shaft(1);
    }
    dir = !dir;
    last_time = ms;
  }
}
```

wehibeg, 2018-12-01:
Thanks for replying

I am using Atmega328p MCU

I need to change the STEP and DIR and PDU , EN pins to match the MCU
correct.

Those pins assigned to a different MCU or CPu board