teemuatlut/TMC2208Stepper #24: Is it possible to use 1 uart

ShohninDmitriy, 2019-02-03:
Is it possible to use 1 uart and switch driver chips using addressing individual pins

[image]

teemuatlut, 2019-02-03:
Possible: quite likely.
Supported by the library itself: no.

You could however define each driver instance to have the same serial port or sw serial pins and then just toggle correct the addressing pin before trying to communicate with a driver.

ShohninDmitriy, 2019-02-03:
а как это реализовать к примеру в marlin or Klipper

aguaviva, 2023-01-29:
What is the reason why the 'addressing' is hard coded? Would it be easy to be able to change the address? That way it could support different addresses so all drivers share the same uart.