teemuatlut/TMC2208Stepper #31: Get actual current micro-step

djairline, 2020-09-03:
Hi !
Thank you very much for this library ! It works amazingly well and is really easy to use.
I am using the tmc2208 to control my stepper on arduino (no 3D printer here :p).
I would like to get the current position between my fullsteps but I can't get driver.cur_a() or driver.MSCNT() to work, when I print the result it always give me zero (even when I am not expecting to be on a fullstep)

Here is my code :

// Author Teemu Mäntykallio, 2017-04-07

// Define pins
#define EN_PIN    5  // LOW: Driver enabled. HIGH: Driver disabled
#define STEP_PIN  13  // Step on rising edge
#define RX_PIN    8  // SoftwareSerial pins
#define TX_PIN    9  //
uint16_t nbStep = 0;
#include <TMC2208Stepper.h>

// Create driver that uses SoftwareSerial for communication
TMC2208Stepper driver = TMC2208Stepper(RX_PIN, TX_PIN);
bool intpolvar = false;
bool dir = true;

void setup() {
  Serial.begin(250000);
  while(!Serial);
  driver.beginSerial(115200);
  // Push at the start of setting up the driver resets the register to default
  driver.push();
  // Prepare pins
  pinMode(EN_PIN, OUTPUT);
  pinMode(STEP_PIN, OUTPUT);

  driver.pdn_disable(true);     // Use PDN/UART pin for communication
  driver.I_scale_analog(false); // Use internal voltage reference
  driver.rms_current(500);      // Set driver current = 500mA, 0.5 multiplier for hold current and RSENSE = 0.11.
  driver.mstep_reg_select(true);
  driver.microsteps(16);
  driver.toff(2);               // Enable driver in software
  
  driver.intpol(intpolvar);
  uint32_t drv_status;
  driver.DRV_STATUS(&drv_status);
  Serial.print("drv_status=");
  Serial.println(drv_status, HEX);

  switch (driver.test_connection()) {
    case 1: Serial.println("Connection error: F");
    case 2: Serial.println("Connection error: 0");
    default: Serial.println("Connection OK");
  }

  digitalWrite(EN_PIN, LOW);    // Enable driver in hardware
}

void loop() {
  uint32_t test;
  digitalWrite(STEP_PIN, HIGH);
  delayMicroseconds(1000);
  driver.MSCNT(&test);
  digitalWrite(STEP_PIN, LOW);
  delayMicroseconds(1000);
  if(nbStep == 155){
    nbStep = 0;
    Serial.println(test);
    Serial.println(driver.cur_a());
  }
  else{
    nbStep++;
    }
}

Thank you very much for your help

Best

closed 2020-09-03 by djairline

teemuatlut, 2020-09-03:
Glad to hear you solved it before I could even read the notification email!

Closed #31<https://github.com/teemuatlut/TMC2208Stepper/issues/31>.

PeterOFre, 2021-03-18:
Hi djairline,
could you please share your connection schematic. I am using an ESP32 together with an TMC2208 module from bigtreetech  and can run it fine in pin mode but want to use spreadcycle mode for more speed. 
So I need to learn how to use the UART mode. Any help is highly appreciated.
best regards
Peter

teemuatlut, 2021-03-18:
https://learn.watterott.com/silentstepstick/configurator/#hardware-connection
https://github.com/bigtreetech/BIGTREETECH-TMC2208-V3.0/blob/master/Hardware/TMC2208connect.png
https://github.com/bigtreetech/BIGTREETECH-TMC2208-V3.0/blob/master/TMC2208-V3.0%20manual.pdf

You need to connect your serial Transmit line to Receive line with a 1kOhm resistor and the Receive line to PDN_UART pin on the driver.

[image]

PeterOFre, 2021-03-18:
Thank you very much for that quick answer, Teemuatlut!
It is is working and I am very happy about that. There are still open questions like how can I switch between StealthChop and SpreadCycle. In your example it is not possible for the TMC2208. I am currently even do not know what is the actual mode. I will share my sketch when I have a solution for that. 
thank you again.
Peter